framps/raspiBackup #899: Sent Telegram notifications to threads: add DEFAULT_TELEGRAM_THREADID

alborworld, 2025-07-08:
Currently, it's possible to send notifications using the variables (see [here](https://www.linux-tips-and-tricks.de/en/raspibackupcategorye/636-raspibackup-invocation-syntax-and-options)) in `/usr/local/etc/raspiBackup.conf`:

- DEFAULT_TELEGRAM_TOKEN
- DEFAULT_TELEGRAM_CHATID
- DEFAULT_TELEGRAM_NOTIFICATIONS

It's not allowed, though, to send a notification to a specific Telegram thread (in case of threaded chats).

Please add (optional) variable DEFAULT_TELEGRAM_THREADID.

framps, 2025-07-08:
Please explain your use case you need this feature.

alborworld, 2025-07-08:
🧭 Why this matters for threaded chats

In my homelab, I use a single Telegram group with multiple forum-style threads (e.g. “raspiBackup,” “Plex,” “Watchtower”) to keep alerts organized by service.

Right now, raspiBackup’s Telegram alerts always land in the group’s main chat (“General”). It’d be ideal to send them into a dedicated “raspiBackup” thread using:

`message_thread_id = 899`

✅ Suggested implementation

Update the [configuration file docs](https://www.linux-tips-and-tricks.de/en/raspibackupcategorye/636-raspibackup-invocation-syntax-and-options) and use-case logic so users can optionally define:

```
DEFAULT_TELEGRAM_THREADID=899
```

—then, when sending Telegram notifications, include it:

```json
{
  "chat_id": "...",
  "message_thread_id": DEFAULT_TELEGRAM_THREADID,   # new support
  "text": "✅ raspiBackup job completed!"
}
```

💡 Example user flow
- Create Telegram thread “raspiBackup” in your homelab group, note its thread_id (e.g. 899).
- Add `DEFAULT_TELEGRAM_THREADID=899` to your `/usr/local/etc/raspiBackup.conf`.
- Continue using existing variables (TOKEN, CHATID).
- Telegram messages now go into the specified thread:
```yaml
Group: Homelab
↳ Thread: raspiBackup
  • ✅ Backup succeeded on pi-zero-2 at 2025-07-08 22:00
````
—Instead of cluttering “General.”

📝 TL;DR
Please add optional `DEFAULT_TELEGRAM_THREADID` variable so raspiBackup can send Telegram notifications to a specific thread inside a group. This enhances organization without altering existing chats or configs.

Let me know if I can help test, review, or prepare a pull request. Thanks for considering!

alborworld, 2025-07-08:
Thank you. The code is ready ([this is the commit](https://github.com/alborworld/raspiBackup/commit/c09b7800cd62a09f7e53cb2dc44797b22728e1e5) on the fork), I will have to test it and then create the PR.

alborworld, 2025-07-09:
Created PR [feat(#899): add support for threaded chats in Telegram notifications](https://github.com/framps/raspiBackup/pull/900).