framps/raspiBackup #906: Command not found

bcutter, 2025-07-25:
Backup is successful, it was just today when I started it manually I discovered this on the CLI:

```
/usr/local/bin/raspiBackup: Zeile 2782: a26559f66f08: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2783: 81edfc629a5d: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2784: 2aa56c490b38: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2785: 1a1c4a3e68de: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2786: d9015b80cd1a: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2787: 8f20dacc79d0: Kommando nicht gefunden.
```

Version:

```
raspiBackup.sh V0.7.1 - 2025-06-11 (cd37c5f)
```

bcutter, 2025-07-25:
I only stop services prior the backup. I noticed those hash-like strings are originating from that stop services section:

```
20250725-131721 DBG 3670:              --> executeBeforeStopServices 
20250725-131721 DBG 3681:              <-- executeBeforeStopServices 
20250725-131721 DBG 3648:              --> stopServices 
07-25-2025 13:17:21 
20250725-131721 DBG 3655:                  --- service nextcloud-news-updater stop; sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --on; docker stop 2c5b844060d0
20250725-131721 DBG 3655:                  --- a26559f66f08
20250725-131721 DBG 3655:                  --- 81edfc629a5d
20250725-131721 DBG 3655:                  --- 2aa56c490b38
20250725-131721 DBG 3655:                  --- 1a1c4a3e68de
20250725-131721 DBG 3655:                  --- d9015b80cd1a
20250725-131721 DBG 3655:                  --- 8f20dacc79d0; systemctl stop docker.service; systemctl stop docker.socket; service cron stop; service redis-wallabag stop; service redis-nextcloud stop; service webmin stop; service nginx stop; service mysql stop; service php8.3-fpm stop; service smbd stop; service minidlna stop
20250725-131721 DBG 2779:                  --> executeShellCommand service nextcloud-news-updater stop; sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --on; docker stop 2c5b844060d0
20250725-131721 DBG 2779:                  --> a26559f66f08
20250725-131721 DBG 2779:                  --> 81edfc629a5d
20250725-131721 DBG 2779:                  --> 2aa56c490b38
20250725-131721 DBG 2779:                  --> 1a1c4a3e68de
20250725-131721 DBG 2779:                  --> d9015b80cd1a
20250725-131721 DBG 2779:                  --> 8f20dacc79d0; systemctl stop docker.service; systemctl stop docker.socket; service cron stop; service redis-wallabag stop; service redis-nextcloud stop; service webmin stop; service nginx stop; service mysql stop; service php8.3-fpm stop; service smbd stop; service minidlna stop
Maintenance mode enabled
2c5b844060d0
/usr/local/bin/raspiBackup: Zeile 2782: a26559f66f08: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2783: 81edfc629a5d: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2784: 2aa56c490b38: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2785: 1a1c4a3e68de: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2786: d9015b80cd1a: Kommando nicht gefunden.
/usr/local/bin/raspiBackup: Zeile 2787: 8f20dacc79d0: Kommando nicht gefunden.
Warning: Stopping docker.service, but it can still be activated by:
  docker.socket
20250725-131740 DBG 2786:                  <-- executeShellCommand 0
```

I think I found the root cause (a recent change when implementing and stopping/starting docker containers and docker services too), but not sure how to fix it:

```
DEFAULT_STOPSERVICES="service nextcloud-news-updater stop; sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --on; docker stop $(docker ps -q); systemctl stop docker.service; systemctl stop docker.socket; service cron stop; .........."
```

I think it breaks at `docker stop $(docker ps -q)` which first queries for all running containers and then stops them. Running `sudo docker ps -q` now gives:

```
2c5b844060d0
a26559f66f08
81edfc629a5d
2aa56c490b38
1a1c4a3e68de
d9015b80cd1a
8f20dacc79d0
```

7 containers where 6 are running. Matches the line errors. So the question is: how to stop docker containers in a way raspiBackup's stop services section can handle it (I don't understand why it can't currently)?

framps, 2025-07-25:
> So the question is: how to stop docker containers in a way raspiBackup's stop services section can handle it

I don't have any docker container up and running and no experience how to stop them.

Please check whether [user provided extensions for docker](https://github.com/framps/raspiBackup/tree/master/extensions_userprovided) helps.

> I don't understand why it can't currently

`docker stop $(docker ps -q)` returns the process list which have to be processed in a loop and passed to the stop command. 

You don't have to add all stop commands in one line. You also can create a shell script which executes all these commands and in there use a loop to stop all containers. Then add this shell script instead of all the commands in the config file. Or you use the user provided plugins to stop and start your containers instead of `docker stop $(docker ps -q)`.

framps, 2025-07-25:
> Is there a way to sort extensions?

No. The extension points are fixed. Looks like an extension FINAL_EXTENSION you're looking for.

As a workaround you could removed the extension logic from the docker extensions and add the scripts to BEFORE_STOPSERVICES and AFTER_STARTSERVICES.

bcutter, 2025-07-25:
It's not perfect this way, but it should work out. I'll monitor the next regular scheduled raspiBackups and likely this issue then can be closed. Thanks a lot especially for pointing to the existing user provided docker extensions.

framps, 2025-07-26:
Yes, this was a very useful extension from @Springjunky .

framps, 2025-07-28:
> sudo nano /usr/local/etc/raspiBackup.conf
--> Removal of all docker-commands in sections "DEFAULT_STOPSERVICES" and "DEFAULT_STARTSERVICES"
--> "docker" added in "DEFAULT_EXTENSIONS" section

Hans-Sh, 2025-07-28:
Thanks!  Portainer stacks are stopped now and backup is started. Works as designed now :-)