framps/raspiBackup #916: stopServices no longer called before smart recycling

framps, 2025-08-08:
### Discussed in https://github.com/framps/raspiBackup/discussions/915

<div type='discussions-op-text'>

<sup>Originally posted by **Markus1406** August  7, 2025</sup>
Hi,

I’ve been using raspiBackup for a long time, and I noticed a behavioral change that seems to have happened in a recent version.

Previously, the backup procedure ran like this:
	1.	Stop all Docker containers
	2.	Create the backup
	3.	Start all Docker containers
	4.	Perform smart recycling
	5.	Send notification email (“: Backup finished successfully”)

Now the order appears to be:
	1.	Stop all Docker containers
	2.	Create the backup
	3.	Perform smart recycling
	4.	Start all Docker containers
	5.	Send notification email (“: Backup finished successfully”)

This change leads to a significantly longer downtime for my Docker services because the smart recycling step takes quite some time. In the past, the containers were already back up and running while recycling was still happening, and the email arrived much later. Now the containers don’t restart until everything is done.

It seems like this behavior has changed with one of the last updates. Is this change documented or configurable? I’m using the GUI to configure raspiBackup and invoke it with raspiBackup.sh.

Thanks in advance!
Markus</div>

Markus1406, 2025-08-09:
@framps 
For me, it’s not that urgent, so you can take your time.
But if you want, you can post the fix, and I can try it out.
Cheers,
Markus

framps, 2025-08-09:
Ok. I created a fix for you 😉  Please execute 
```
curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_916
```
to download the fix.

I suggest to use for your test a different backup location than the one you use normally - just to make sure your existing backups are not affected in case of any serious bug in the fix. Or even better execute the test on a non production system if one is handy.

closed 2025-08-14 by framps