framps/raspiBackup #923: Add zstd support for faster compression

tbertels, 2025-09-25:
[Zstandard](https://en.wikipedia.org/wiki/Zstd) aka zstd is faster and has a slightly better compression ratio than gzip with default settings.

Here's a comparison for total backup time with a Raspberry Pi 4 with a compressed tar file sent to a NAS.

* gzip:
```
real    36m11,688s
user    0m0,005s
sys     0m0,011s
```

Compressed filesize: 7,2 GiB (7 781 091 436)

* Zstandard:
```
real    23m22,393s
user    0m0,007s
sys     0m0,010s
```

Compressed filesize: 7,0 GiB (7 484 355 114)

With `tar`, the parameter to use Zstandard compression is `--zstd`

framps, 2025-09-25:
Sounds interesting 👍 

Found these interesting pages:

https://linuxreviews.org/Comparison_of_Compression_Algorithms

https://jothiprasath.com/blog/gzip-vs-zstd/

You're looking for fast compression speed. Others are looking for optimal compression. Next are looking for fast decompression. Every tool has it's pros and cons. 

So from my point of view it doesn't make sense just to add zstd support. As far as I understand right now for tar it's quite easy to pass the compression tool to use as a parameter. tar decompression will use the required compression tool.

I will do some evaluation on this compression and check what's the effort to add tar compression options to select the compression tool used by tar. I will not cover dd.

tbertels, 2025-10-05:
I've tested it with `sudo ./raspiBackup.sh -t tar --tarCompressionTool zstd` and it worked fine, creating a `.zst` file instead of `.tgz` for the main partition.

Note that the `.zst` file and its folder still contain "tgz" in their names even though `.tar.zst` is used.
The compressed file should ideally be named `.tar.zst` to reflect that it's a zst compressed tar file. This also makes it possible to see the actual files when we open it, instead of having to first extract the tar file.

Then I've compared with `sudo diff --brief --recursive` the files from the `.zst` file with the files from a mounted offline .img file backup done with `ddrescue`.
Apart from a few missing or different files due to the difference in backup times and offline vs online backup, they were identical.

I didn't try restoring the backup as my current card reader or card adapter isn't very reliable. I already had problems with ddrescue.

tbertels, 2025-10-06:
> > The directory name will be kept to be tgz to indicate it's a compressed tar backup.
> 
> This unfortunately will require a lot of changes in the code because of option `-z` which will be kept for backward compatibility. Therefore the directory name will contain `tar` even it's a compressed directory.

I think that's actually better than `tgz` for the non-gzip compression options.

> The backups now end with `.tar.zst` as suggested 😄

Nice, thanks!